LilyGO/TTGO-TS #1: MPU9250

donesnowangel, 2018-02-09:
Hi, I'm trying to work with the MPU9250 onboard the board with 1.8 inch screen. Can you please give me some guidance as to how to use it? What pins it is on? Is it on I2C or SPI? I've tried using various sketches examples and libraries, but without any success. An i2c scanner says there are no devices and when I try to get an a WHO_AM_I, I get a 0xFF. 

Thanks,
Daniel

circuitsforfun, 2018-02-18:
Same problem here, I have both the 1.4" and 1.8" screen version and can't get the MPU9250 to work.
It would be great to see some example code provided or the schematic so I can see how it is connected.

donesnowangel, 2018-02-18:
I actually got it to work. Apparently the IMU uses pins 19 and 18 for SDA and SCL. So if you use the standard Wire.h library, add Wire.begin(19,18);

Daniel

circuitsforfun, 2018-02-18:
Ahh yes, that did the trick... thank you!

BeatArnet, 2018-05-19:
Hi, Could you please post an arduino example with the MPU9250. Thanks a lot.

LilyGO, 2018-12-19:
https://github.com/LilyGO/TTGO-TS-V1.2/blob/master/MPU9/MPU9.ino

closed 2018-12-19 by LilyGO

kotakomputer, 2019-06-20:
TTGO V2.1_1.6 does not have PIN 18. Any recomended PIN?