LilyGO/TTGO-TS #3: button example - strange behavior

GittaHuby, 2018-02-20:
The button example does not work properly: If i press btn A down it works as expected. While releasing btn A the serial monitor writes 'btn B released'. Physically buttons B and C without any function. Any idea?

kosso, 2018-03-12:
Through some trial and error I have discovered the v1.2 board (with gyro) has these pin numbers for the three pins (using the same orientation as the photo) :  
~~~
LEFT 35
MIDDLE 34 
RIGHT 39
~~~

Note: Some of the examples use M5Stack.h (from a very similar board. Is this a dev version?) which defines BUTTON_A_PIN 39, BUTTON_B_PIN 38 and BUTTON_C_PIN 37 which are wrong (at least for the V1.2 board containing the gyro and fewer physical pins).

This is why the example doesn't work.

jvhaarst, 2018-06-02:
This works for me : 
```
#include <RBD_Timer.h>  // https://github.com/alextaujenis/RBD_Timer
#include <RBD_Button.h> // https://github.com/alextaujenis/RBD_Button

// input pullup enabled by default
RBD::Button left_button(35);
RBD::Button middle_button(34);
RBD::Button right_button(39);

void setup() {
  Serial.begin(115200);
}

void loop() {
  if(left_button.onPressed()) {
    Serial.println("Left Button Pressed");
  }

  if(left_button.onReleased()) {
    Serial.println("Left Button Released");
  }
  if(middle_button.onPressed()) {
    Serial.println("Middle Button Pressed");
  }

  if(middle_button.onReleased()) {
    Serial.println("Middle Button Released");
  }
  if(right_button.onPressed()) {
    Serial.println("Right Button Pressed");
  }

  if(right_button.onReleased()) {
    Serial.println("Right Button Released");
  }
}
```

GittaHuby, 2018-06-10:
Perfect ! Many Thanks, now it also works for me.
best regards

closed 2018-06-10 by GittaHuby