LilyGO/TTGO-TS #4: MySD?

Galbi3000, 2018-03-08:
In the SD example you #include a header file called mySD.h but you do not include this library anywhere in the Git project.

According to your device diagram and looking at the ESP32__goodSD.ino source code I see that you have connected the SD card in a non-standard way for the ESP32. For example you use IO15 for MOSI but the standard SD library for ESP32 uses the designated MOSI pin of the ESP32 (IO23) so it would seem your custom mySD library is required to make the SD card work on your device!

Where can I find this library?

kosso, 2018-03-09:
Same issue. 
The file `mySD.h` does not exist.

kosso, 2018-03-09:
I see the files required exist in this library: https://github.com/nhatuan84/esp32-micro-sdcard 

But I still cannot get it to work. 

Anyone have any idea what the correct pin settings are for the `SD.begin()` statement?

ririfonfon, 2018-03-11:
SD.begin(13,15,2,14) with V1.0 it work.

ririfonfon, 2018-03-11:
SD.begin(13,15,2,14) 
13=SD3-CS
15=CMD-MOSI
2=SDO-MISO
14=CLK-SCK

kosso, 2018-03-12:
@ririfonfon Thanks!  That worked on my V1.2.

Now I need the pin numbers for the 3 buttons. Any idea?  Thanks.

kosso, 2018-03-12:
Update: It's OK. I've discovered through some trial and error that the 3 button pins are (in the orientation of the photo):
~~~
LEFT 35
MIDDLE 34 
RIGHT 39
~~~

(Note: The examples use `M5Stack.h` (from a very similar board. Is this a dev version?) which defines `BUTTON_A_PIN 39`, `BUTTON_B_PIN 38` and `BUTTON_C_PIN 37`  which are wrong (at least for the V1.2 board containing the gyro and fewer physical pins)

ririfonfon, 2018-03-12:
yes i use RBD_Button  https://github.com/alextaujenis/RBD_Button
35 34 39 
but if i use the m5stack.h the screen don't work
the screen in the lib. is ILI9341 and on the ttgo ts v1.0 is ST7735

do you have another lib. for the speaker ?
or the gpio ?

kosso, 2018-03-12:
For the screen, I found the Adafruit_ST7735 library worked : 

```
#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library
#include <SPI.h>

#define TFT_CS   10
#define TFT_RST  9  
#define TFT_DC   8
#define TFT_SCLK 13   
#define TFT_MOSI 11

//Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_MOSI, TFT_SCLK, TFT_RST);
Adafruit_ST7735 tft = Adafruit_ST7735(16, 17, 23, 5, 9); // CS,A0,SDA,SCK,RESET

float p = 3.1415926;

void setup(void) {
  Serial.begin(115200);
  Serial.print("Hello! ST7735 TFT Test");

  // Use this initializer if you're using a 1.8" TFT
  tft.initR(INITR_BLACKTAB);   // initialize a ST7735S chip, black tab

  Serial.println("Initialized");

  //tft.setRotation(0); [0-3]

  tft.fillScreen(ST7735_BLACK);

  delay(500);

  // large block of text
  tft.fillScreen(ST7735_RED);
  tft.setTextColor(ST7735_WHITE);
  tft.setTextSize(1);
  tft.setCursor(2, 2);
  
  // Wrap some text, changing the colour of the words on the way. 
  tft.setTextWrap(true);
  testdrawtext("hello world! ", ST7735_WHITE);
  testdrawtext("hello world! ", ST7735_BLACK);
  testdrawtext("hello world! ", ST7735_WHITE);
  testdrawtext("hello world! ", ST7735_BLACK);
  testdrawtext("hello world! ", ST7735_WHITE);
  testdrawtext("hello world! ", ST7735_BLACK);
  testdrawtext("hello world! ", ST7735_WHITE);
  testdrawtext("hello world! ", ST7735_BLACK);
  testdrawtext("hello world! ", ST7735_WHITE);
  testdrawtext("hello world! ", ST7735_BLACK);
  testdrawtext("hello world! ", ST7735_WHITE);
  testdrawtext("hello world! ", ST7735_BLACK);
  testdrawtext("hello world! ", ST7735_WHITE);
  testdrawtext("hello world! ", ST7735_BLACK);
  testdrawtext("hello world! ", ST7735_WHITE);
  testdrawtext("hello world! ", ST7735_BLACK);
  testdrawtext("hello world! ", ST7735_WHITE);
  testdrawtext("hello world! ", ST7735_BLACK);
  testdrawtext("hello world! ", ST7735_WHITE);
  testdrawtext("hello world! ", ST7735_BLACK);
  testdrawtext("hello world! ", ST7735_WHITE);
  testdrawtext("hello world! ", ST7735_BLACK);
  testdrawtext("hello world! ", ST7735_WHITE);
  testdrawtext("hello world! ", ST7735_BLACK);
  testdrawtext("hello world! ", ST7735_WHITE);
  testdrawtext("hello world! ", ST7735_BLACK);
  testdrawtext("hello world! ", ST7735_WHITE);
  testdrawtext("hello world! ", ST7735_BLACK);
  testdrawtext("hello world! ", ST7735_WHITE);
[... 206 lines cut ...]
}

void mediabuttons() {
  // play
  tft.fillScreen(ST7735_BLACK);
  tft.fillRoundRect(25, 10, 78, 60, 8, ST7735_WHITE);
  tft.fillTriangle(42, 20, 42, 60, 90, 40, ST7735_RED);
  delay(500);
  // pause
  tft.fillRoundRect(25, 90, 78, 60, 8, ST7735_WHITE);
  tft.fillRoundRect(39, 98, 20, 45, 5, ST7735_GREEN);
  tft.fillRoundRect(69, 98, 20, 45, 5, ST7735_GREEN);
  delay(500);
  // play color
  tft.fillTriangle(42, 20, 42, 60, 90, 40, ST7735_BLUE);
  delay(50);
  // pause color
  tft.fillRoundRect(39, 98, 20, 45, 5, ST7735_RED);
  tft.fillRoundRect(69, 98, 20, 45, 5, ST7735_RED);
  // play color
  tft.fillTriangle(42, 20, 42, 60, 90, 40, ST7735_GREEN);
}
```

I'm only just getting around to looking at the speaker stuff.

ririfonfon, 2018-03-12:
yes me too the lib. adafruit work
i have the buttons, screen, sd. but no speaker

kosso, 2018-03-12:
In the TEST_speaker folder, did you update your pin button definitions in `Config.h`  https://github.com/LilyGO/TTGO-TS-V1.2/blob/master/TEST_speaker/lib/src/utility/Config.h 

The speaker test uses the incorrect button definitions. But you should still automatically hear a short bit of audio when it starts up.

kosso, 2018-03-12:
@ririfonfon Try my test sketch here, where I have extracted the relevant bits code from the M5 library test :  https://github.com/kosso/ESP32_TTGO_TFT_SD_Sketches/tree/master/TTGO_Speaker

ririfonfon, 2018-03-12:
good it's work
thank you

ririfonfon, 2018-03-12:
all fonction work
.beep();
.tone(x,y);
.update();
thank you very much, nice job

Galbi3000, 2018-05-06:
I have v1.2 with the 1.88 TFT.

When I did get it working using the ST7735 libraries I found the screen did not display correctly. The graphics demo for TTGO worked but the display was shifted one pixel left and one pixel up with random pixels in the unused edges. I left it at that for a while until I bought 2 screens for other Arduino devices.

While getting those to work I found an excellent TFT library elsewhere on GitHub that is faster than the Adafruit ones and it supports many TFT driver chips in one library.

I decided to try that library with the TTGO and got the same problem of shifted display with random pixels. However, that library has much more support for ST7735 variants, by selecting a non-standard variant I was able to make the screen display correctly.

The library I used is called TFT_eSPI at https://github.com/Bodmer/TFT_eSPI
To get it to work with my TTGO I made the following changes in the User_Setup.h file:

- Make sure it's using #define ST7735_DRIVER
- Make sure TFT_WIDTH and TFT_HEIGHT are 128 and 160
- THIS IS THE IMPORTANT BIT! Use #define ST7735_GREENTAB2
- Use the following values for the ESP32 Dev Board settings:
  #define TFT_MISO 19
  #define TFT_MOSI 23
  #define TFT_SCLK 5
  #define TFT_CS 16
  #define TFT_DC 17
  #define TFT_RST 9

The #define ST7735_GREENTAB2 setting corrected the shifted display without altering the colours. The default is ST7735_BLACKTAB.
With those changes to User_Setup.h the TTGO screen works perfectly and is much faster than the Adafruit libraries :)

BorisFR, 2018-05-07:
I have found the same info for screen, except I  put RST to -1.
Do you not if it is possible to put the screen backlight off?

jvhaarst, 2018-06-02:
For me the default library works, using this code : 
```
#include "SD.h"
#include "SPI.h"

#define PIN_NUM_MISO 2
#define PIN_NUM_MOSI 15
#define PIN_NUM_CLK  14
#define PIN_NUM_CS   13

void setup() {
    Serial.begin(115200);
    SPI.begin(PIN_NUM_CLK, PIN_NUM_MISO, PIN_NUM_MOSI, PIN_NUM_CS);
    if(!SD.begin(PIN_NUM_CS)){
        Serial.println("Card Mount Failed");
        return;
    }
    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);
}
void loop() {}
```

ydsdsy, 2018-08-02:
Hi,
Update : 
I found this code and with a little tweaking it came to work for me : 
[https://github.com/LilyGO/TTGO-TS-V1.2/blob/master/TEST_SD/ESP32__goodSD.ino](url)
i can correctly see the sdcard and the files on it, can write and read the files....
it all came down to this settings : SD.begin(23,5,19,18)

complete working code below : 
` // include the SD library:
#include <SPI.h>
#include <mySD.h>

File root;

void setup()
{
  Serial.begin(115200);
  Serial.print("Initializing SD card...");
  /* initialize SD library with SPI pins */
 if (!SD.begin(23,5,19,18)) {            //T1:13,15,2,14  T2: 23,5,19,18
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");
  /* Begin at the root "/" */
  root = SD.open("/");
  if (root) {    
    printDirectory(root, 0);
    root.close();
  } else {
    Serial.println("error opening test.txt");
  }
  /* open "test.txt" for writing */
  root = SD.open("test.txt", FILE_WRITE);
  /* if open succesfully -> root != NULL 
    then write string "Hello world!" to it
  */
  if (root) {
    root.println(" Dog : Luna : 0,128s : 4,058s : 4,186s : ");
    root.flush();
   /* close the file */
    root.close();
  } else {
    /* if the file open error, print an error */
    Serial.println("error opening test.txt");
  }
  delay(1000);
  /* after writing then reopen the file and read it */
  root = SD.open("test.txt");
  if (root) {    
    /* read from the file until there's nothing else in it */
    while (root.available()) {
      /* read the file and print to Terminal */
      Serial.write(root.read());
    }
    root.close();
  } else {
    Serial.println("error opening test.txt");
  }
  
  Serial.println("done!");
}

void loop()
{}

void printDirectory(File dir, int numTabs) {
  
  while(true) {
     File entry =  dir.openNextFile();
     if (! entry) {
       break;
     }
     for (uint8_t i=0; i<numTabs; i++) {
       Serial.print('\t');   // we'll have a nice indentation
     }
     // Print the name
     Serial.print(entry.name());
     /* Recurse for directories, otherwise print the file size */
     if (entry.isDirectory()) {
       Serial.println("/");
       printDirectory(entry, numTabs+1);
     } else {
       /* files have sizes, directories do not */
       Serial.print("\t\t");
       Serial.println(entry.size());
     }
     entry.close();
   }
}`

Yves,

LilyGO, 2018-12-19:
https://github.com/nhatuan84/esp32-micro-sdcard

closed 2018-12-19 by LilyGO

bnmahajan78, 2019-01-06:
This approach  worked form me .

#include "FS.h"
#include "SD.h"
#include "SPI.h"

#define PIN_NUM_MISO 2
#define PIN_NUM_MOSI 15
#define PIN_NUM_CLK  14
#define PIN_NUM_CS   13

 
 
void setup(){
    Serial.begin(115200);
    
   // if(!SD.begin()){
   //     Serial.println("Card Mount Failed");
   //     return;
   // }

    SPI.begin(PIN_NUM_CLK, PIN_NUM_MISO, PIN_NUM_MOSI, PIN_NUM_CS);
    if(!SD.begin(PIN_NUM_CS)){
        Serial.println("Card Mount Failed");
        return;
    }

    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);
    uint8_t cardType = SD.cardType();

    if(cardType == CARD_NONE){
        Serial.println("No SD card attached");
        return;
    }

    Serial.print("SD Card Type: ");
    if(cardType == CARD_MMC){
        Serial.println("MMC");
    } else if(cardType == CARD_SD){
        Serial.println("SDSC");
    } else if(cardType == CARD_SDHC){
        Serial.println("SDHC");
    } else {
        Serial.println("UNKNOWN");
    }

    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);

   
    Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
    Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));
}

void loop(){

}