LilyGO/TTGO-TS #10: TFT 1.44 V1.0 White screen

skelstar, 2018-07-07:
I bought a v1.0 1.44inch module. Out of the box it has what looks like the example app listed in this repo.

The code has: ```Adafruit_ST7735 tft = Adafruit_ST7735(16, 17, 23, 5, 9); // CS,A0,SDA,SCK,RESET''' 
...and looks like the pins that I should be using, but after writing to the chip I'm getting a blank screen.

I have a ```Serial.println("Ready...");``` which is being executed, so the chip was porgrammed ok.
Am I missing something?

Thanks for your help in advance!
Sean

themakerbro, 2019-02-18:
Did you ever solve this?

skelstar, 2019-04-05:
> 
> 
> Did you ever solve this?

I did... not sure what I did different, but this is just setup that I think was working:
`Adafruit_ST7735 tft = Adafruit_ST7735(16, 17, 23, 5, 9); // CS,A0,SDA,SCK,RESET`