LilyGO/TTGO-TS #19: Battery voltage

dastrobbel, 2020-01-21:
I want to read the battery lvl of a lipo 3.7V battery and need to no what pin it uses. I already read the pins from 0 to 40. But none gave the right value.

sfambach, 2020-05-21:
Hi, I tried to find out if there is a pin already assigned over a resitance bridge or something else but i found no circuit diagram. Would it at least be possible to post a full ciruit diagram instead of the shrinked one that is on git. Or is it possible to do something over the vbus?
Thank in advanced for you reply SFA

G6EJD, 2020-12-09:
There is no on-board voltage divider. You have to wire one externally from Vbat to A0 (goio17)

Konjohn, 2022-10-05:
Hi, Can someone solve this problem?