LilyGO/TTGO-TS #22: TTGO TS I²C communication address

yusufCHE, 2021-04-16:
Hi
how can I find the adresse of TTGO TS used in I²C communication?

sfambach, 2021-04-17:
Hi, i dont know if I get the question. Which I2C address do you mean, if you use it as a client you can set the address. If you need an address of peripherie there are some scanner for wire bus on the net or use the examples.
If this does not help, please clarify you needs.

yusufCHE, 2021-04-17:
I want to use it as a slave and the arduino as master.
I used the code scanner with arduino but it doesn't work. I didn't use pullup resistors so I suppose that its is integrated in the TTGO TS.
Thank you

sfambach, 2021-04-17:
Then you need to create a programm  for the TS and let it work as a I2C client. You can choose an adress for it. The basis of the TS, is  an ESP32 and its programmable with the arduino gui. Or you put an AT Firmware on the  ts and configure it with at commands. Be carful dependant on which arduino you use, you need an logic level converter from 5V to 3.3V.

yusufCHE, 2021-04-17:
So I don't have to use wire library to programme the TS, because in this library to assign the address to the Arduino UNO, for example, we have to write it as an argument of the function wire. Begin () which is different from the ESP32 board that uses (SCL, SDA) pins as arguments.
How can I assign an address to the ESP32 board ?
THANK YOU

sfambach, 2021-04-17:
Hi, sorry I was not aware that arduino does not support client mode for ESP32. I Thought this works as on arduino. Maybe you can follow this link: https://github.com/gutierrezps/ESP32_I2C_Slave.
BR