LilyGO/TTGO-TS #23: Screen won't turn on!

MaxSMoke777, 2021-05-14:
The unit I got from Amazon worked fine with it's pre-installed examples. but I haven't been able to get the screen to turn no matter what sketch or settings I use. The one exception is when I tried an M5 example, which causes the screen solid white. Otherwise it doesn't seem to respond to anything. 

Is it possible I accidentally pushed some series of buttons that turns off the screen at a firmware level? If the screen was broken, it shouldn't be able to show white, right?

sfambach, 2021-05-14:
Hi, it is some time ago when i tested this device, I cant remember if there were specific problems. Ther are at least two different versions, but the settings (pins) for the displays looks the same.  Here is my test project you can have a look at the pin settings : https://github.com/sfambach/esp32/tree/master/project/TtgoTsJPegViewer. Here are some additional tests I made https://github.com/sfambach/esp32/tree/master/boards/TTGO_TS_V1.2_MPU9250.  If you post your code i can check it with my device if you want. BR Stef

MaxSMoke777, 2021-05-14:
Thanks for responding! I found out what the issue was and posted the fix in the Amazon review for anyone else unlucky enough to buy this janky board. The issue is that it's back-light has to be turned on by switching pin 27 to HIGH. Failure to do so means the screen appears broken. The board actually has a TFT_eSPI profile which does not, itself, have the setting to turn on the back-light. 

It also didn't help that the board's name and appearance is alot like the TTGO TS, when it's really the T10-V2.0 board. They look identical in photos, but actually have separate libraries. It was when I found the right library, loaded an example that actually worked (many in the package actually don't), I noticed the back-light switch in setup and started adding it in. Adding that to the other examples fixed everything.

andrademarcos27, 2021-05-30:
@MaxSMoke777 , you said you found the right library. I am facing the same problem that you described earlier. Could you please share where you found a library that works?

MaxSMoke777, 2021-05-30:
> @MaxSMoke777 , you said you found the right library. I am facing the same problem that you described earlier. Could you please share where you found a library that works?

https://github.com/Xinyuan-LilyGO/LilyGo_Txx
(and don't forget to set 27 HIGH, or the back-light won't come on and it'll still look dead)

andrademarcos27, 2021-05-30:
Thanks @MaxSMoke777. Now it´s alive and working. Finally !!!